davilla/atvusb-creator #640: where is the program for mac

peterblabla, 2015-05-22:
i tried a version that would run but would not recognize my usb thumb drive any help?

peterblabla, 2015-05-24:
still no help fuck were are you

thibmaek, 2017-08-22:
You can compile it yourself from the atvusb-creator directory in this repository. Also change your tone. In 2015 you were asking for somethings someone else made for free and had at that point long been deprecated.

thibmaek, 2018-08-10:
@rsspdx hmm I suspect this will only work with strict requirements (Install Python2.5, PyQt4, and py2app).

Could you past the error output?

rsspdx, 2018-08-14:
Thank you.  Can’t install Python 2.5 on my Mac. I’ll wait until I have a Linux box to try it on.

===
Richard Sherman 
rss.pdx@gmail.com