davilla/atvusb-creator #645: Does not work anymore?

AlbRipi, 2018-09-19:
Hallo, I had lost my old patchstick and it seems I can't create a new one anymore. Mac version exit with error "IndexError: list index out of range". Windows version gives the error "please check internet connection". So it seems the software it's trying to access something that isn't available anymore. Can you suggest a way around? Or anybody has a disk-image of a working patchstick? Thank you.

teknofile, 2018-11-21:
Would be interested if someone found a way around this.

amet, 2019-02-09:
@teknofile @AlbRipi 

https://github.com/davilla/atvusb-creator/releases/tag/1.0.b13

ping me if you run into any issues... I have an emotional attachment to this project :)

teknofile, 2019-05-21:
Will do! May be a couple weeks before this project gets back up to the top of my radar.

AlbRipi, 2019-05-25:
Ok, I can open the App but does not recognize any USB pendrive. I've tried 3 USB pendrive: 512MB, 4GB, 32GB and from both Mavericks and SnowLeopard, but nothing.

amet, 2019-05-28:
Are you running it using sudo commands I posted?

amet, 2019-05-28:
And only tested on macOS Mojave , no guarantees for earlier versions

teknofile, 2019-06-21:
@amet 

thanks! now that I'm off of holiday, I had a few moments to pick this up and start playing. Was able to get it to download and make the patch stick. Next steps for me will be to get linux installed on it... I dont need/want to run frontrow on it. again, Thanks!

amet, 2019-06-21:
Best thing in my opinion would be to create osmc install stick that essentially installs stripped down version of Linux with Kodi on it. Then disable Kodi on start and you will have a Linux server