davilla/atvusb-creator #646: Trying to hack my Apple TV 1Gen. Need help…

Agelandkaai, 2019-03-08:
Hi, i found atvusb-creator-1.0.b13, but it gives an error when I try to open it.

An unexpected error has occured during execution of the main script. 
OSError: [Errno 30] Read-only file system: '/private/var/folders/2I/2_jg_h7n5kndwxqwyk4dgf7m0000gn/T/AppTranslocation/BD2D7F55-0930-47BB-94D9-6A279B395CAA/d/staging'

My iMac runs 10.13.2

I bought a 1gen Apple TV to use it as a hard disk, since I have no wifi in my laundryroom. 
We want to watch some tv while ironing…

If someone can help us, we would be very gratefull.
Also maybe a nice tutorial? Not really good at Terminal stuff, copy and paste I can do, but don't want to mess up my iMac.

Thank you

amet, 2019-03-08:
https://github.com/davilla/atvusb-creator/releases/tag/1.0.b13

You need to run sudo command as per the note on the release. That’s the only way it can be done unfortunately :(

Agelandkaai, 2019-03-08:
Is that then to make the patchstick?
btw, thank you for ur swift response.

Agelandkaai, 2019-03-08:
Thank you very much, got it to open.
For other people who are (like me) a bit afraid to use the terminal,
I opened it with Show Package, after left clicking on the file.
Then clicking Contents, Mac OS --> ATV usb creator.

amet, 2019-03-08:
Good to know, let us know if you run into any other issues

Agelandkaai, 2019-03-09:
yeah, well, the program keeps saying there is no usb drive to be found.

amet, 2019-03-09:
How big is the usb stick and can you paste the output from terminal when you run the sudo

Agelandkaai, 2019-03-09:
128 gb

Agelandkaai, 2019-03-09:
on my laptop now, can paste it later

amet, 2019-03-09:
128 should work, we accept all up to 256.

Agelandkaai, 2019-03-09:
Worked with an old ipod shuffle 1GB.
How do I get to watch movies from on the hard drive. without needing wifi?

Agelandkaai, 2019-03-09:
Spoke too soon
[image]

amet, 2019-03-09:
its important to run the command I linked you to, it has to run as sudo for it to be able to create the  patchstick

`sudo /atvusb-creator.app/Contents/MacOS/atvusb-creator

e.g: MacBook-Pro-3:~ amet$ sudo /Users/amet/Desktop/atvusb-creator.app/Contents/MacOS/atvusb-creator`

Agelandkaai, 2019-03-09:
So I open the terminal, and just copy the command in to it?
I'll try

Agelandkaai, 2019-03-09:
Command not found
Where does the file need to be? Now its in Downloads.

amet, 2019-03-09:
not really :)

type sudo and "space" ,  find atvusb-creator.app Show contents and find atvusb-creator, you can drag/drop atvusb-creator into terminal window.. it would look like below 

```sudo /Users/amet/Desktop/atvusb-creator.app/Contents/MacOS/atvusb-creator```

Agelandkaai, 2019-03-09:
[image]
So i typed sudo & space, then dropped the app on the terminal and then pressed Enter, got this

Agelandkaai, 2019-03-09:
after typing my password, that is

amet, 2019-03-09:
you need to read what I posted :)
```sudo /Users/iMac/Downloads/atvusb-creator.app/Contents/MacOS/atvusb-creator```

Agelandkaai, 2019-03-09:
okay, dropped the app from the package, got this far
now running the script again

Agelandkaai, 2019-03-09:
Lasting longer already now

Agelandkaai, 2019-03-09:
copying files to target disk

Agelandkaai, 2019-03-09:
copy files to target disk
  remove recovery hfsplus GUID
  umount_disk settling delay (10 seconds)
  change recovery to real GUID
Unable to add real recovery: gpt add: unable to open device '/dev/disk3': No such file or directory
  install_recovery failed
Failed! (0:04:13)

Agelandkaai, 2019-03-09:
don't see the usb in my Finder sidewindow, or with disk utility

Agelandkaai, 2019-03-09:
the ipod is still flashing it's little orange light

amet, 2019-03-09:
You will have to find another usb to use, I am not sure it will work with an iPod

Agelandkaai, 2019-03-09:
the only other one i have is the 128, which he doesn't see, or an even older ipod shuffle, the "long" white one, this is the silver one with a clip

amet, 2019-03-09:
Patchstick creator needs to partition usb into 2 separate partitions, not sure that’s possible with iPod ... I have not tried it so can’t offer any advice on it

Agelandkaai, 2019-03-09:
With the 128 Gb, he keeps saying 
Unable to find any USB drives

Agelandkaai, 2019-03-09:
And may I ask, what does Bootloader do?

amet, 2019-03-09:
It’s all part of “jailbreaking” the atv1 and opening it up to be seen as a storage drive and media player.

How you planning to watch movies from it? Not much software will run on it

Agelandkaai, 2019-03-09:
I actually just want to use the hd, where i would store the files and want to watch these, since we don't have wifi in the laundryroom.

amet, 2019-03-09:
Usb you are trying to create will only open atv1 for you to use terminal and add software to it, it will not have any additional software in it for you to watch anything

Agelandkaai, 2019-03-09:
But it's possible to add software, to watch files from the hard drive?

Think I still have a usb stick on my sailing boat that stores music.
Maybe I'll go and get it later today.

Agelandkaai, 2019-03-09:
Is it difficult to add the software? Otherwise I bought this one with no reason…

Agelandkaai, 2019-03-09:
Fingers crossed, went for a normal usb-stick on the boat.

Agelandkaai, 2019-03-09:
Okay, have the patch-stick.

Agelandkaai, 2019-03-09:
Is it normal the apple tv doesn't show up in my iTunes? Have iTunes 12.8.…

Agelandkaai, 2019-03-09:
Airplay works, but don't get a list of devices

amet, 2019-03-09:
Yes it normal, Apple dropped support for it long time ago

Nothing we worked on in 2009-10 still runs on that hardware :(

As far as I can remember, osmc.tv is the only player solution , not sure how it works or if you can get movies on it.

https://osmc.tv/wiki/apple-tv/frequently-asked-questions/

amet, 2019-03-09:
Last time I used atv1 was approximately 9 years ago, since then many things have changed.

Remember that if you install osmc it will delete all content from that appletv and instal Linux on it, just a warning :)

Agelandkaai, 2019-03-09:
So now I restarted the Apple tv, with the usb in it.
It says if no errors, plug in your aple tv, wait 3 minutes, and open Tewrminal.
How do i plug my apple tv in to my iMac? ethernet?

amet, 2019-03-09:
Apple TV needs to be on Wifi or on LAN

But I still fail to see what you will do after you successfully ran a patchstick on it... it will give you a terminal access and a new menu... that’s it

Menu is no good as all the apps that were in it are long gone

Agelandkaai, 2019-03-09:
He's on the same wifi as my iMac.
Typed ssh frontrow@appletv.local  enter
nothing happened

So I can't then just use the HD to store files, and play them without using iTunes?
That was the reason I bought this one.

Agelandkaai, 2019-03-09:
And I can't use it anymore either with iTunes?
Wath was the last supported version of iTunes?

amet, 2019-03-09:
> He's on the same wifi as my iMac.
> Typed ssh [frontrow@appletv.local](mailto:frontrow@appletv.local) enter
> nothing happened
> 
> So I can't then just use the HD to store files, and play them without using iTunes?
> That was the reason I bought this one.

no, there is no way for you to use the HD to store files and play them using ATV1 player

amet, 2019-03-09:
> And I can't use it anymore either with iTunes?
> Wath was the last supported version of iTunes?

not sure what version was last that supported it, if I had to guess.. 5 years ago or so

Agelandkaai, 2019-03-09:
Anyway, a very big Thank You for what you've helped me with already.

amet, 2019-03-09:
👍